Howchoo/pi-power-button #3: Use another GPIO (not GPIO3)

matiasz8, 2018-05-04:
Hello,

I'm testing this project. I have a problem.

Your example works ok for me. 

In this moment I need to use the GPIO3, for the touchscreen.

So, Is it possible to use another gpio, not the gpio3 in this case?

Thanks!

Nizhile, 2019-04-17:
You can use any GPIO available to shutdown the raspberry. Simply update `listen-for-shutdown.py` to change GPIO number.
Anyway, only GPIO#3 on pin 5 will allow you to wake on your raspberry PI as described in [Pi power button guide](https://howchoo.com/g/mwnlytk3zmm/how-to-add-a-power-button-to-your-raspberry-pi).

ZLevine, 2019-07-22:
@Nizhile is correct -- pin 5 (GPIO3) is necessary for waking the Pi. If you don't care about this functionality and only want a shut**down** button, you can use a different pin. But if you want it to power on too, that's the only pin you can use unfortunately.

closed 2019-07-22 by ZLevine