Howchoo/pi-power-button #5: pull up resistor

Konsts, 2018-12-23:
I get that error on mi Pi 3B:
`RuntimeWarning: A physical pull up resistor is fitted on this channel! GPIO.setup(3, GPIO.IN, pull_up_down=GPIO.PUD_UP)`

Nizhile, 2019-04-17:
According to https://pinout.xyz/pinout/pin5_gpio3# GPIO3 already includes a build-in pull up for I2C clock. 
> SCL (i2c Clock) includes a fixed, 1.8 kohms pull-up to 3.3v

Thus GPIO library displays this warning, but there's no effect on your reset button. Problem will occur only if you need to use GPIO3 without the pull-up.

Anyway, I found some reset button schema that explicitly includes this pull-up as a resistor between Pin#1 (3v3) and Pin#5 (GPIO3). Maybe their rapsberry pi pull-up is not present or does not work as expected.

ZLevine, 2019-07-22:
This is a warning, not an error -- it won't affect anything. Closing.

closed 2019-07-22 by ZLevine