Howchoo/pi-power-button #8: RPi.GPIO not installed on OSMC

JohnFitz, 2019-05-14:
I liked the article so thought I'd add a switch to OSMC running on my RPi3. Unfortunately, it doesn't seem to have the RPi.GPIO module installed by default. 

I got it working by following [this post](https://discourse.osmc.tv/t/solved-cannot-install-rpi-gpio/71906), but I'm new to this so I can't verify that everything in the post is really necessary. If somebody knows more, perhaps it could be added to the installation script.

JohnFitz, 2019-05-14:
Just noticed that one of the commenters on the accompanying article wrote:

> Very good Post.
> I found in my pi3 that RPi.GPIO were not loaded.
> To fix it I did this:
> 
> sudo apt-get install gcc python-dev python-pip i2c-tools libi2c-dev python-smbus
> sudo pip install RPi.GPIO

[Link to original comment](http://disq.us/p/1pzmg51)

ZLevine, 2019-07-22:
This looks correct to me -- if the latest version of OSMC still doesn't have RPi.GPIO installed, the commands above should install it. If you're unsure, try backing up your Pi first ([Windows](https://howchoo.com/g/nmexndnlmdb/how-to-back-up-a-raspberry-pi-on-windows) or [Mac](https://howchoo.com/g/ztqymgezm2u/create-a-backup-image-of-your-raspberry-pi-sd-card-in-mac-osx) and giving it a shot.

I wouldn't worry about any of those packages being unnecessary -- they're unlikely to conflict with existing OSMC packages, and will likely just use up a _tiny_ bit of extra space at best.

Just eyeballing them, these do look like packages you'd want to install to properly interact with the GPIO interface.

closed 2019-07-22 by ZLevine