Howchoo/pi-power-button #9: No keyboard and no ssh connection available

codegastudio, 2019-06-10:
Hi,
I follow your tuto and when the service is started, i can't use the keyboard anymore and can't access to the pi with ssh to.
Any idea ?
Best regards

ZLevine, 2019-07-22:
This sounds like it's likely unrelated. It's possible something got corrupted along the way.

Can you try reinstalling Raspbian run through the steps again? You can back up your Pi's SD card if you'd like to preserve it before trying (instructions on [Windows](https://howchoo.com/g/nmexndnlmdb/how-to-back-up-a-raspberry-pi-on-windows) or [Mac](https://howchoo.com/g/ztqymgezm2u/create-a-backup-image-of-your-raspberry-pi-sd-card-in-mac-osx).

KonradLinkowski, 2021-09-03:
> Hi,
> I follow your tuto and when the service is started, i can't use the keyboard anymore and can't access to the pi with ssh to.
> Any idea ?
> Best regards

Any update on that @codegastudio