Howchoo/pi-power-button #10: Adding 'listen-for-shutdown' code kills the HifiBerry Audio card

PaulM47, 2019-07-17:
Adding your 'listen-for-shutdown' code kills my HifiBerry audio card, it was working before adding the code, not working afterwards :-(

I assume it's some clash of resources, any suggestions please ??

ZLevine, 2019-07-22:
Which HiFiBerry exactly do you have? I imagine what's happening is your HiFiBerry uses pin 5 (SCL). See the [HiFiBerry GPIO history](https://www.hifiberry.com/build/documentation/gpio-usage-of-hifiberry-boards/). Some of them state the following:

> On the HiFiBerry Digi+ Pro GPIOs **5** and 6 are also use and cannot be used for anything else.

> GPIO2-3 (pins 3 and **5**) are used by our products for configuration. If you are experienced with I2C, you might add other slave devices.

Can you confirm which version your HiFiBerry is?

PaulM47, 2019-07-22:
Thank you for replying. I have the DAC+ board, and yes it appears to be the SCL pin, GPIO3, that is the problem. It's a pity this pin is hard wired for power up, but I guess it's a convenient side effect rather than a design feature. Unfortunately the case I'm using has only one front panel button so I'll have to use it for power up and use the software to power down.
Maybe worth a note in your article warning that the 'listen-for-shutdown' code may clash with boards that use the I2C Interface.

closed 2019-07-22 by PaulM47