Howchoo/pi-power-button #16: Possible to use other pins?

janis-ax, 2020-03-05:
Is it possible to use another pin rather Pin5 ?

ZLevine, 2020-03-05:
Unfortunately not for full functionality. There are two main features of the power button here:

1. Shut the Pi down on press (the Pi now consumes zero power)
2. Turn the Pi on (wake the Pi) on press

The wake functionality (#2) requires the SCL pin, pin 5. So if you don't care about turning the Pi back on using the power button, you could use a different GPIO pin for #1 above. Then, to turn the Pi back on, you'd need to disconnect and reconnect the power cord (or use a cord with a physical switch in it).

Note: Of course, for the other connection, you can use any ground pin you want.

I hope this was helpful. Let me know if you have any other questions.

closed 2020-03-05 by ZLevine

seltix5, 2020-09-22:
> Unfortunately not for full functionality. There are two main features of the power button here:
>
> 1. Shut the Pi down on press (the Pi now consumes zero power)
> 2. Turn the Pi on (wake the Pi) on [...]

this info should be on the README.md, thanks !

ZLevine, 2020-09-22:
@seltix5 thanks for the suggestion—I'll add it now. :)