Howchoo/pi-power-button #18: Other GPIOs?

koko004, 2020-06-12:
Is it possible to use other GPIOs? I already have those pins busy with a HAT 0.96 OLED screen. Thanks for the project it is brilliant and it works perfect.

ZLevine, 2020-06-12:
Unfortunately, not for 100% functionality. While you can use any other pin for GND, the wake functionality relies on pin 5 (SCL). In order to consume zero power while in the off state, this pin shorting to GND is a sort of "hardware" method of waking the Pi.

So if SCL is in use, you could use an alternate GPIO pin to _shut down_ the Pi, but you wouldn't have wake functionality. If you did this, you'd need to unplug the Pi and plug it back in to turn it on.

Are you 100% sure your screen is _using_ SCL and not just covering it up? [Pinout.xyz](https://pinout.xyz/) is a good resource for identifying which pins are in use.

closed 2020-06-12 by ZLevine

koko004, 2020-06-15:
Thanks for your response, bit I think that it´s used
[image]
[image]