Howchoo/pi-power-button #28: -bash: ./pi-power-button/script/install: File or folder does not exist.

nmbrg, 2022-01-01:
Cloning the repo:
git clone https://github.com/Howchoo/pi-power-button.git
[image]

Running the following gets an error saying that the file or folder does not exist:
./pi-power-button/script/install
[image]
(unfortunately the text is in swedish, so ive translated it in the title of the issue)

looking in the folder with "ls" i can see the install file:
[image]

nmbrg, 2022-01-02:
For some reason it worked today on both my raspberry pi's.

Did not change a thing, in fact ive not touched the raspberry pi in any way since it i created this issue.
Well well.. you can close this.

prstrm, 2022-01-05:
Hej nmbrg!

From what I can see from your second screenshot it seems you already are inside of the` $/pi-power-button/script/`-directory when running the install-command. If I remember correctly dot-slash `./` in the beginning of the command tries to run the command from where currently are and in this case it wants to run `~/pi-power-button/script/pi-power-button/script/install` which does not exist. While in that directory `./install` would have done it I believe. At least I think this is the case here.

nmbrg, 2022-01-05:
> Hej nmbrg!
>
> From what I can see from your second screenshot it seems you already are inside of the` $/pi-power-button/script/`-directory when running the install-command. If I remember correctly [...]

you are probably correct. thanks for pointing that out. now atleast i know what to check for the next time i get an error like that trying to run something on my pi. :)