Howchoo/pi-power-button #37: Not working under Bookworm?

JoeClueless, 2024-05-12:
This worked fine for me for several years, but now - after a fresh installation of BOOKWORM, it fail.
The Shutdown-Functionality does not work any longer.  I made a fresh Bookworm installation from scratch on a Pi Zero W,
installed the howchoo-script (installation script finished with success message), powered the PI Zero of via SSH 'sudo poweroff'.
I can restart the Pi with the button, but I can't power it off...
Any idea for me noob?

winslowj, 2024-06-27:
I can’t get to shutdown or wake pi 4b

bassamanator, 2024-07-02:
It's not working under RPi OS Bookworm (`cat /etc/debian_version`...`12.6`).

I caught the exception, `Failed to add edge detection`. The [RPi.GPIO package](https://pypi.org/project/RPi.GPIO/) has not changed in ages, so I can't quite tell what the issue is 😕.

bassamanator, 2024-07-02:
This functionality was broken due to a recent linux kernel update. Read about it [here](https://pypi.org/project/rpi-lgpio/).

## Solution

You simply need to install `rpi-lpgio`, which will automatically uninstall `rpi-gpio`. [Source](https://rpi-lgpio.readthedocs.io/en/latest/install.html).

```shell
sudo apt update && sudo apt install python3-rpi-lgpio
```

jimholmesNZ, 2024-07-16:
I installed the rpi-lpgio, do I need to change the listen-for-shutdown.py script to look at rpi-lpgio instead of rpi-gpio? Cheers Jim.

bassamanator, 2024-07-16:
> I installed the rpi-lpgio, do I need to change the listen-for-shutdown.py script to look at rpi-lpgio instead of rpi-gpio? Cheers Jim.

No other changes are necessary.

jimholmesNZ, 2024-07-16:
Sorry, I didn't see your reply and changed my original comment. I still can't get the shutdown to work after running the 
sudo apt update && sudo apt install python3-rpi-lgpio command.

bassamanator, 2024-07-16:
@jimholmesNZ Have you restarted the service, or rebooted?

jimholmesNZ, 2024-07-16:
Yes, I've tried both rebooting and restarting the listener. I'm just about to ditch Bookworm and go back to Bullseye!

bassamanator, 2024-07-16:
@jimholmesNZ To downgrade the entire OS for this script unwise, but maybe you have a special use case?

Feel free to install [my fork](https://github.com/bassamanator/rpi-power-button) of this script, install it via the instructions (via systemd).

jimholmesNZ, 2024-07-16:
Thanks very much, using your fork and following the instruction it now works, it seems that it needs a long press on the switch? Anyway thank you so much, now I just need to work out a command to turn on / off the monitor with Bookworm and I'm just about one. Thanks, Jim.

winslowj, 2024-07-16:
I just ground out the pi using the 3v pin. Probably not the best but clearly there’s no good alternative solution for buttons on RPi without hacking into the OS. You’d think after all these years there would be a standard pin added for buttons

bassamanator, 2024-07-16:
> I just ground out the pi using the 3v pin. Probably not the best but clearly there’s no good alternative solution for buttons on RPi without hacking into the OS. You’d think after all these years there would be a standard pin added for buttons

Not sure what device you're using, but no `hacking into the OS` is required.

winslowj, 2024-07-16:
> Not sure what device you're using, but no `hacking into the OS` is required.

I’d surely prefer not to have to ground out the pi, but nothing I’ve tried here works with RPi 4b bookworm

bassamanator, 2024-07-16:
@winslowj What is the issue exactly? There are `ground` pins btw.

hackslikeus, 2025-01-12:
Bookworm is now suited for virtual environment.   See this [tutorial](https://learn.adafruit.com/python-virtual-environment-usage-on-raspberry-pi/overview) for more info.  for me running dietpi, quick fix - i added path to my virtual environment in the shell script.
Was:  # Carry out specific functions when asked to by the system
case "$1" in
  start)
    echo "Starting listen-for-shutdown.py"
    /usr/local/bin/listen-for-shutdown.py &
    ;;
Now:  # Carry out specific functions when asked to by the system
case "$1" in
  start)
    echo "Starting listen-for-shutdown.py"
    **/home/pi/path-to/venv/** /usr/local/bin/listen-for-shutdown.py &
    ;;